ucsdsysnet/sparsdr #1: Compressing PlutoSDR Source block?

gefa, 2019-08-26:
Could you share the PlutoSDR source block too? Even if it isn't complete, I could use it to build my own or improve it. I do not have N210, and B210 requires expensive license, so I' left with PlutoSDR to implement sparSDR.

samcrow, 2019-10-03:
The GNU Radio integration for the PlutoSDR is different from the USRP version in a way that makes it difficult to get compressed samples into GNU Radio. From what I remember, some part of the system interprets the compressed samples as numbers and converts them to floating-point. It should be possible to disable this, but we have not yet found an easy way.

We showed that the compression on the PlutoSDR works, but we have not yet spent the time to get it working with GNU Radio. Unfortunately, this block does not exist yet, even in an incomplete form.

dkozel, 2020-03-29:
The gr-iio driver for the PlutoSDR has an internal `short to complex float 32bit` conversion. You can either use the `fmcomms2 Source` which will give you the raw compressed samples or I'd be happy to work with you to update the `Pluto Source` to have the option to select which output data type is used.

closed 2021-10-07 by samcrow

dkozel, 2021-10-08:
Thanks @rsubbaraman and @samcrow!

As an FYI, work is just happening this week to simplify the PLUTO SDR and general gr-iio blocks in GNU Radio. As part of that the direct passthrough of the data from the FPGA should work in the next few days.

The related work is at:
https://github.com/gnuradio/gnuradio/pull/5147