ucsdsysnet/sparsdr #3: Reconstruct block does not delete temporary directory

samcrow, 2020-01-21:
The SparSDR reconstruct block creates a `sparsdr_reconstruct_[various characters]` folder somewhere every time it runs. It should delete the folder when it is destructed, but it currently does not.

samcrow, 2020-01-30:
New discovery: When using the block from GNU Radio Companion, pressing the stop button in the toolbar causes the process to exit without deleting the folder.
When using the block from a Python script run from the command line, pressing control-C twice causes the process to exit and delete the folder as it should.

dkozel, 2020-03-29:
We've made some fixes to the way flowgraphs are stopped in GNU Radio since you identified this issue. They're merged into the maint-3.8 and master branches, so they aren't immediately accessible to you since the gr-sparsdr OOT hasn't been updated for the 3.8 release. If updating is of interest here's a link to the migration guide.
https://wiki.gnuradio.org/index.php/GNU_Radio_3.8_OOT_Module_Porting_Guide