ucsdsysnet/sparsdr #12: Error during 'make' step for gr-sparsdr

tkpudgy, 2021-10-18:
Hello,

I've followed the steps in this link sequentially. All went smooth until the 'make' step for gr-sparsdr:

https://github.com/ucsdsysnet/sparsdr/blob/pluto/doc/getting_started.md

Any help or workaround would be appreciated.

Thank you,

Jae

[ 37%] Building CXX object lib/CMakeFiles/test-sparsdr.dir/test-sparsdr_autogen/mocs_compilation.cpp.o
[ 40%] Building CXX object lib/CMakeFiles/test-sparsdr.dir/test_sparsdr.cc.o
[ 42%] Building CXX object lib/CMakeFiles/test-sparsdr.dir/qa_sparsdr.cc.o
[ 45%] Linking CXX executable test-sparsdr
/usr/bin/ld: libgnuradio-sparsdr-1.0.0git.so.0.0.0: undefined reference to `gr::filter::rational_resampler_base_ccf::make(unsigned int, unsigned int, std::vector<float, std::allocator<float> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [lib/CMakeFiles/test-sparsdr.dir/build.make:157: lib/test-sparsdr] Error 1
make[1]: *** [CMakeFiles/Makefile2:285: lib/CMakeFiles/test-sparsdr.dir/all] Error 2
make: *** [Makefile:160: all] Error 2

samcrow, 2021-10-19:
It looks like the problem is a version mismatch. The gr-bluetooth module in the `pluto` branch currently works with GNU Radio 3.7 and needs a few changes to support GNU Radio 3.8.

I made a version that should work with GNU Radio 3.8, which is in the `pluto-3.8` branch. Please try that and let me know if there are other problems.

samcrow, 2021-10-22:
It's possible that the files ended up installed in different places on your computer. When you run `make install` in the gr-sparsdr build folder, it will print the path to each installed file.

On my computer, that process installs the files `/usr/local/lib/python3/dist-packages/sparsdr/__init__.py` and `/usr/local/lib/x86_64-linux-gnu/libgnuradio-sparsdr.so.1.0.0git` (as well as various other files). If those files are in different places on your computer, you will need to change the environment variables to point there.

As you can tell, I barely spent any time testing gr-sparsdr with GNU Radio 3.8 on other computers. This process is not going very smoothly, but we will get through it and make the system work.

tkpudgy, 2021-10-30:
Those warnings and errors disappeared once I made changes to the reconstruct block. I'm not seeing anything in the waterfall graph but I suppose it's because there is no real-world signal that it's capturing that meets the threshold? I do need to to familiarize myself with gnuradio and the sparsdr blocks. Any suggestion on setting up a quick demo to show end-to-end capability would be great. Thank you so much for all your help!

samcrow, 2021-11-01:
We were missing an actual working example for the Pluto version. I added a flow graph file `SparSDR Pluto single channel.grc` and some documentation to [the new examples folder](https://github.com/ucsdsysnet/sparsdr/tree/b9ed92a73a3193f381d38258d1d656d8ff03d99d/gr-sparsdr/examples). That flow graph receives one BLE advertising channel, reconstructs the signals, and displays them in a waterfall view.

Let me know if you have any problems with that, or if you would like help adapting it to do something else.

tkpudgy, 2021-11-02:
Thank you so much for the example. I just downloaded the file and tried to open it on gnuradio. However, there's a long list of errors and doesn't open. I couldn't capture the beginning of the error. I just took a snapshot and pasted below. Any help would be appreciated. 

/home/pi/sparsdr/gr-sparsdr/examples/SparSDR Pluto single channel.grc:2063:13:FATAL:PARSER:ERR_GT_REQUIRED: Couldn't find end of Start Tag a line 2060
/home/pi/sparsdr/gr-sparsdr/examples/SparSDR Pluto single channel.grc:2065:11:FATAL:PARSER:ERR_TAG_NAME_MISMATCH: Opening and ending tag mismatch: div line 1904 and a
/home/pi/sparsdr/gr-sparsdr/examples/SparSDR Pluto single channel.grc:2141:145:FATAL:PARSER:ERR_ATTRIBUTE_WITHOUT_VALUE: Specification mandate value for attribute data-pjax
/home/pi/sparsdr/gr-sparsdr/examples/SparSDR Pluto single channel.grc:2145:24:FATAL:PARSER:ERR_ATTRIBUTE_WITHOUT_VALUE: Specification mandate value for attribute data-pjax
/home/pi/sparsdr/gr-sparsdr/examples/SparSDR Pluto single channel.grc:2145:24:FATAL:PARSER:ERR_SPACE_REQUIRED: attributes construct error
/home/pi/sparsdr/gr-sparsdr/examples/SparSDR Pluto single channel.grc:2145:24:FATAL:PARSER:ERR_GT_REQUIRED: Couldn't find end of Start Tag a line 2145
/home/pi/sparsdr/gr-sparsdr/examples/SparSDR Pluto single channel.grc:2152:15:FATAL:PARSER:ERR_TAG_NAME_MISMATCH: Opening and ending tag mismatch: div line 2138 and a
/home/pi/sparsdr/gr-sparsdr/examples/SparSDR Pluto single channel.grc:2171:160:FATAL:PARSER:ERR_ATTRIBUTE_WITHOUT_VALUE: Specification mandate value for attribute preload
/home/pi/sparsdr/gr-sparsdr/examples/SparSDR Pluto single channel.grc:2173:124:FATAL:PARSER:ERR_ATTRIBUTE_WITHOUT_VALUE: Specification mandate value for attribute data-close-dialog
/home/pi/sparsdr/gr-sparsdr/examples/SparSDR Pluto single channel.grc:2241:14:FATAL:PARSER:ERR_ATTRIBUTE_WITHOUT_VALUE: Specification mandate value for attribute disabled
/home/pi/sp

samcrow, 2021-11-02:
Those errors look like the software was parsing HTML when it expected flow graph XML. You may have downloaded the GItHub web page that displays the .grc file, instead of downloading just the .grc file. Does something different happen if you download the file from this raw link: <https://github.com/ucsdsysnet/sparsdr/raw/b9ed92a73a3193f381d38258d1d656d8ff03d99d/gr-sparsdr/examples/SparSDR%20Pluto%20single%20channel.grc> ?

tkpudgy, 2021-11-08:
Thank you for all your help so far. I was able to get the waterfall graphs to display. I think at this point I just need to familiarize myself with SparSDR capability. Thank you!

tkpudgy, 2021-12-04:
Some questions about the "Bin specification" parameter on the "Compressing Pluto Source" block: 

In the example grc file, this is set to 416..450:12800. 1) How do you calculate which frequency bins 416 through 450 are? 2) All other frequency bins other than 416..450 are masked off meaning turned off, correct?

Thank you.

samcrow, 2021-12-04:
#### Bin to frequency conversion

There are two steps to find the corresponding frequency range:

1. Convert bin numbers from FFT order to frequency order

In the bin specification, bin numbers use the order produced by the FFT. Bins 0 through 511 correspond to the upper half of the frequency range, and bins 512 through 1023 correspond to the lower half of the frequency range. Bin 0 is the center frequency, and bin 1023 is just below the center frequency.

To do the conversion: for each bin number that is less than 512, add 512. If the bin number is greater than or equal to 512, subtract 512.

This will give bin numbers that are proportional to frequency. Some contiguous ranges in FFT order will be split into two separate ranges in frequency order.

In this example, bins 416..450 are all less than 512. Adding 512 to each gives the bin range 928..962 .

2. Convert bin numbers to frequency

SparSDR on the Pluto uses 1024 bins that span 61.44 MHz of frequency. First calculate the frequency of bin 0: `bin_0_frequency = center_frequency - 61.44 MHz / 2`

Then, use that to find the frequency of each bin: `bin_frequency = bin_0_frequency + bin_number * 61.44 MHz / 1024`

In this example, the center frequency is 2.4 GHz, so bin_0_frequency is 2.36928 GHz. Bin 928 corresponds to 2.42496 GHz, and bin 962 corresponds to 2.427 GHz. This is range of approximately 2 MHz centered at 2.426 GHz, one of the BLE advertising channels.

I know this is not very user-friendly, but we are working on a simpler way to configure these settings.

#### Other bins

Yes, all bins that are not mentioned in the bin specification are masked, so the FPGA will not send any samples for those bins.

mkhazraee, 2021-12-04:
Actually there is a python script that does this computation:
https://github.com/ucsdsysnet/sparsdr/blob/universal/examples/uhd_rx_compressed_cfile/fft_bin_calculator.py

Just update the 3 values for Pluto as mentioned in line 7. The input to the script is 
<capture center freq> <band center freq> <bandwidth>
The idea is you give it the capture center frequency, and the center frequency of the band you want, with band bandwidth. It will output the FFT bin ranges, and the actually frequencies that are going to be captured (after proper rounding). 

If you want multiple bands, simply run it for different bands, and use them all for capture. (For reconstruct they should be processed separately). 

We will soon incorporate this script into our GnuRadio module.

tkpudgy, 2021-12-04:
Thank you Sam and mkhazree. I understand the bin number to frequency translation now. Would you please elaborate on the "Shift amount" parameter and on what circumstances this needs to be adjusted? Another questions about the "Band x bins" parameter on the "SparSDR Reconstruct" block: Is my understanding correct in that this number is bigger than the number of bins specified in the Source block (i.e. 34) to ensure entire coverage of the bandwidth being captured? Thank you.

mkhazraee, 2021-12-05:
1) You can think of shift amount as a knob for digital gain. 

When we do 1024 point FFT on 13-bit inputs, the output will be 13+10+1=24 bits. 13 bits assuming there is IQ imbalance correction causing one additional bit to 12-bit DAC, that 10 comes from stages and there is a corner overflow case that adds another bit. On the other hand we backhaul 16-bit values from Pluto. So we have to shift the outputs by 24-16=8 bits to the right keeping the most significant bits, which is the default shift value. 

However this is the most conservative value possible. In practice that overflow never happens, as it requires perfect sine. Also the implementation on Pluto actually doesn't add any bits in the IQ imbalance step, so it stays at 12 bits (just wanted to be conservative for future designs as it barely changed the resource utilization). So choosing 6 instead of 8 is safe too. Now if you want to get digital gain, you can go for lower values. For example, if the receive antenna is not that close to the the transmitter the value 4-5 worked really nice for us, aka 2-4 times digital gain, on top of 4 times safe gain.

Note that the threshold is applied after the shift, on 16-bit values, so you should update the thresholds accordingly. 

2) One beauty of being in the frequency domain is that we can decide on the time-domain sample rate later. For example, each bin is 60KHz, and if I have 10 bins active, I'm not obliged to get an 600KHz time domain signal. Instead I can assume 40 bins of zero, and get 3MHz time domain signal. Generally we benefit from frequency domain sparsity, so based on the desired sample rate, we fill the rest of the non-active bins with 0 and then we perform the IFFT.

If there is a single band, number of reconstruct bins should be at least number of capture bins, not to lose any information, and can be more to get a nice time-domain sample rate to be fed to the decoder. If there are adjacent bins, let's say capturing 2 adjacent Bluetooth bands, actually we can give each 2MHz to a different reconstruct path, and fill each of them with extra zeros if we want a different sample rate, let's say that 3MHz as example above.

tkpudgy, 2021-12-10:
Thank you mkhazrazee. I have a better understanding of the shift amount and the flexibility to resample at a different rate going into time-domain. 

As a first order demo, I would like to show my colleagues the advantage of using SparSDR by making the comparison of file size of saved data between SparSDR vs stock Pluto. Before heading in that direction, I thought I would experiment with SparSDR alone and show the file size difference when choosing a different threshold. This will demonstrate how SparSDR essentially chooses to keep signals that are significant, dictated by a threshold parameter. So for the BLE band we've been discussing so far, I varied the threshold from 100 to 50000. If my understanding is correct in that lower number for this parameter means lower threshold, I would expect to see a bigger file size for the saved data as more signals would be deemed significant. In fact for roughly about 25 seconds of data collection for each threshold, I saw 150 MB (100) vs 60 MB (50000). What puzzled me was the waterfall plot that I was seeing. Please see the snapshot I took in the attachment. I was expecting to see more sparse and muted plot for the threshold of 50000, and more noisy and intense plot for the 100. But it was the opposite. Any explanation you can provide would be helpful. Also, if you have a better and simple way to do setup a quick demo to show advantage of using SparSDR, I would much appreciate it. Thank you!

[image]
[image]

samcrow, 2021-12-11:
Your understanding about the threshold is correct. The reason the waterfall graphs look like that is that when there is a time period with no signals, the reconstruction application does not produce any samples.

Let's look at an example: We capture a signal with a strong Bluetooth signal, a gap of 1 second where there are weak signals and background noise, and finally another strong Bluetooth signal at the end.

With a low threshold, the radio will receive and store the strong Bluetooth signals and the weak signals during the gap. When we reconstruct the signals, we will get approximately one second of weak signals in the gap between the two Bluetooth signals.

With a high threshold, the radio will receive and store the strong Bluetooth signals but ignore the weak signals between them. When we reconstruct the signals, the reconstruction software sees the first Bluetooth signal, a 1-second gap with nothing, and the second Bluetooth signal. Instead of filling the gap with 0 samples, the reconstruction software skips that time period. It reconstructs the two Bluetooth signals with no gap between them, so they look like they are consecutive in time.

In your example with the low threshold, the gaps between the strong signals are periods where there were weak signals. By increasing the threshold, you eliminated those weak-signal periods and the strong signals ended up compressed together in time. That makes the waterfall diagram more crowded, but if you look at the number of samples in your reconstructed files you should see significantly more samples from the lower threshold.

Let me know if you have other questions.

mkhazraee, 2021-12-11:
Actually overlap-gaps branch is exactly to add filling zeros for such examples. Right now it only supports N210, but soon we will port it to Pluto too. Will let you know when it's done, shouldn't take long.

samcrow, 2021-12-11:
The sparsdr_reconstruct application from the [pluto-overlap-gaps branch](https://github.com/ucsdsysnet/sparsdr/tree/pluto-overlap-gaps) now produces zero samples in the gaps, so it should work more like you expect.

stevewillson, 2023-03-09:
The "Bin specification" parsing is done in `compressing_source.cc`

The string format in the "Bin specification" field is as follows:

`BIN_1`:`THRESHOLD_VALUE_1`,`BIN_2`:`THRESHOLD_VALUE_2`,...

Ex:
`2:1000,44:500`

This backhauls bin 2 with threshold value 1000 and bin 44 with threshold value 500.

or:

`START_BIN_1`..`END_BIN_1`:`THRESHOLD_VALUE_1`,`START_BIN_1`..`END_BIN_1`:`THRESHOLD_VALUE_1`,...

Ex:
`2..4:1000,44..55:500`

This backhauls bins 2-4 (inclusive) with threshold value 1000 and bins 44-55 (inclusive) with threshold value 500.