ucsdsysnet/sparsdr #27: 'combined_pluto_receiver_sptr' object has no attribute 'stop_all'

gefa, 2022-04-11:
```
/home/gnuradio/sparsdr/gr-sparsdr/examples/sparsdr_pluto_combined_block.py
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-gnuradio'
Launching reconstruct: /home/gnuradio/.cargo/bin/sparsdr_reconstruct sparsdr_reconstruct --no-progress-bar --log-level INFO --compression-fft-size 1024 --compressed-bandwidth 61.44e6 --sample-format v2 --timestamp-bits 30 --source sparsdr_reconstruct_V94gGl/compressed.pipe --reconstruct-band 64:64:1.39999e+07:sparsdr_reconstruct_V94gGl/0.pipe
gr::log :WARN: file_source0 - file size is not a multiple of item size
Traceback (most recent call last):
  File "/home/gnuradio/sparsdr/gr-sparsdr/examples/sparsdr_pluto_combined_block.py", line 177, in <module>
    main()
  File "/home/gnuradio/sparsdr/gr-sparsdr/examples/sparsdr_pluto_combined_block.py", line 153, in main
    tb = top_block_cls()
  File "/home/gnuradio/sparsdr/gr-sparsdr/examples/sparsdr_pluto_combined_block.py", line 80, in __init__
    self.variable_sparsdr_combined_pluto_receiver_0.stop_all()
AttributeError: 'combined_pluto_receiver_sptr' object has no attribute 'stop_all'
15:27:42 [INFO] Decimation 16
15:27:42 [INFO] Window sum 32
15:27:42 [INFO] Scale 0.0009765625
```
stop_all function is defined in compressing_source.cc - it seems that SWIG didn't run properly. I wonder if it's easier to fix this or port everything to gnuradio 3.9 which uses Pybind11 instead of SWIG.

samcrow, 2022-04-12:
It looks like SWIG wasn't configured to make bindings for the functions of the compressing_source class. I pushed to the master branch commit a27abdc190028a0e5b688ea556f6071e5f1ff42f , which I think fixes that. Let me know if it works.

gefa, 2022-04-12:
Thanks for a quick fix! That makes sense.

closed 2022-04-12 by gefa