ucsdsysnet/sparsdr #29: Pluto works in manual rx gain mode only

gefa, 2022-04-13:
```
/home/gnuradio/sparsdr/gr-sparsdr/examples/sparsdr_pluto_combined_block.py
Launching reconstruct: /home/gnuradio/.cargo/bin/sparsdr_reconstruct sparsdr_reconstruct --no-progress-bar --log-level INFO --compression-fft-size 1024 --compressed-bandwidth 61.44e6 --sample-format v2 --timestamp-bits 30 --source sparsdr_reconstruct_9L8TiQ/compressed.pipe --reconstruct-band 64:64:1.39999e+07:sparsdr_reconstruct_9L8TiQ/0.pipe
gr::log :WARN: file_source0 - file size is not a multiple of item size
Traceback (most recent call last):
  File "/home/gnuradio/sparsdr/gr-sparsdr/examples/sparsdr_pluto_combined_block.py", line 105, in <module>
    main()
  File "/home/gnuradio/sparsdr/gr-sparsdr/examples/sparsdr_pluto_combined_block.py", line 83, in main
    tb = top_block_cls()
  File "/home/gnuradio/sparsdr/gr-sparsdr/examples/sparsdr_pluto_combined_block.py", line 38, in __init__
    self.variable_sparsdr_combined_pluto_receiver_0.set_gain(60)
  File "/usr/local/lib/python3/dist-packages/sparsdr/sparsdr_swig.py", line 2122, in set_gain
    return _sparsdr_swig.combined_pluto_receiver_sptr_set_gain(self, gain)
RuntimeError: Failed to write gain attribute
20:36:25 [INFO] Decimation 16
20:36:25 [INFO] Window sum 32
20:36:25 [INFO] Scale 0.0009765625
```

Thanks for adding "gain control mode option" a few weeks ago. However, I'm having trouble using it.

Any ideas what may be the issue? I would expect the "Gain" (numeric) text field in _Pluto Receive and Reconstruct block_ to disappear  when "Gain control mode" field is anything other than "manual". Maybe it's related to this.

Manual gain mode works fine though. Other options: fast_attack, slow_attack, and hybrid AGC all result in this error. I don't think it's a hardware problem since the AGC feature works with default firmware using PlutoSDR source.

samcrow, 2022-04-14:
Thanks for your detailed report. I updated the block YAML files in the master branch so that the generated code will only set the gain value if the gain control mode is manual. Similarly, the gain field in the user interface now only appears if the gain control mode is manual. That should match what the Pluto source block from gr-iio does.

Let me know if the new version works.

gefa, 2022-04-17:
Thank you for a quick fix! It works now!

closed 2022-04-17 by gefa