ucsdsysnet/sparsdr #34: Ethernet does not work

lobart, 2023-02-10:
Hi
How i can turn on ethernet on sparsdr?

samcrow, 2023-02-11:
Can you please explain more about what you are trying to do?

Are you using a USRP N210 or a Pluto?

lobart, 2023-02-11:
@samcrow hi
I am using Pluto plus

samcrow, 2023-02-12:
That is a surprising problem. Our patches should not change anything about USB or networking.

Does everything work using a normal image on the Pluto?

The pages linked from the Pluto quick start guide might help: https://wiki.analog.com/university/tools/pluto/users/quick_start

lobart, 2023-02-12:
@samcrow 
Yes, i got v0.33 plutosdr-fw and apply patches from https://github.com/plutoplus/plutoplus/tree/master/patches
Then i apply patches from this repo, then i build firmware and copy .frm files to PlutoSDR.
And block compressing pluto source run with next error.
[image]
I attached plutos dmesg and building log vivado.
[vivado.log](https://github.com/ucsdsysnet/sparsdr/files/10715801/vivado.log)
[dmesg.log](https://github.com/ucsdsysnet/sparsdr/files/10715802/dmesg.log)

samcrow, 2023-02-12:
Thanks for the details; this makes more sense now.

I've only tested SparSDR with a normal Pluto and based on the normal firmware from Analog Devices. The Pluto+ looks interesting, but since I've never tried it I may not be able to help very much.

These lines from the dmesg log look important:
```
ad9361 spi0.0: ad9361_dig_tune_delay: Tuning RX FAILED!
cf_axi_adc: probe of 79020000.cf-ad9361-lpc failed with error -5
```

`cf-ad9361-lpc` is the IIO device that software uses to read received samples. That is the same in the normal firmware and the SparSDR firmware. Because it failed to start, the software can't receive any signals.

If you run `iio_info` on the Pluto, it will show you all the IIO devices. I expect that `cf-ad9361-lpc` will be missing.

I would suggest trying this:
1. With the normal Pluto firmware from Analog Devices, see if `cf-ad9361-lpc` gets initialized correctly
1. With the Pluto+ patches but without the SparSDR patches, see if `cf-ad9361-lpc` gets initialized correctly
1. With the Pluto+ and SparSDR patches, see if `cf-ad9361-lpc` gets initialized correctly

That should help tell if there is a hardware problem, a problem with the Pluto+ patches, or a problem caused by both sets of patches.

lobart, 2023-02-13:
@samcrow 
Thanks for your answer!
Results:
1. cf-ad9361-lpc gets initialized.
2. cf-ad9361-lpc gets initialized.
3. I found a vivado compile time error. Then i can't test your third version. 
I think there are conflict with pins mapping in vivado project.
[vivado.log](https://github.com/ucsdsysnet/sparsdr/files/10721734/vivado.log)

lobart, 2023-02-13:
@samcrow 
I applied only hdl patch and built firmware.
eth0 is lost. But cf-ad9361-lpc is initialized.

lobart, 2023-02-13:
@samcrow dmesg logs with sparsdr and plutoplus
[sparsdr_dmesg.log](https://github.com/ucsdsysnet/sparsdr/files/10723816/sparsdr_dmesg.log)

samcrow, 2023-02-13:
It looks like the Pluto+ and SparSDR patch sets have some conflicts.
If you apply the SparSDR HDL patch but not the other SparSDR patches, SparSDR will not work. You need to resolve the conflict that causes the Vivado compile error.

I wish I could help more, but I don't have a Pluto+ device or enough time to solve these problems. Good luck with the debugging.