ucsdsysnet/sparsdr #36: Receive frequencies for USRP N210

stevewillson, 2023-02-15:
I am attempting to update gr-sparsdr for gnuradio version 3.10. The main work involves creating bindings using pybind11 rather than swig. 

I am using a NI USRP-2922 (same as a USRP-N210) and the following gnuradio-companion flow graph:

[image]

I receive the following signals: 
 
[image]

I expect there to be signals received around 2.402, 2.426, and 2.480 GHz (BLE advertising channels). Instead, I see a significant number of received signals around 2.430GHz. 

Are there recommended debugging steps to determine the cause of this issue?

samcrow, 2023-02-16:
This looks like just a minor problem with the flow graph.

The center frequency of the compressed capture is 2.5 GHz (2500 MHz).
Output 0 from the reconstruct block is offset -25 MHz from the center, so its center frequency is 2500 MHz - 25 MHz = 2475 MHz.
The left waterfall view (I think that's connected to output 0) shows frequent signals 5 MHz above its center frequency, which is really 2480 MHz (one of the advertising channels).
Output 1 is offset 25 MHz from the center, so its center frequency is 2500 MHz + 25 MHz = 2525 MHz.

If you change the center frequency of the compressing USRP source to 2.45 GHz, the other frequencies will be accurate and the signals should look correct.

stevewillson, 2023-02-16:
> This looks like just a minor problem with the flow graph.
>
> The center frequency of the compressed capture is 2.5 GHz (2500 MHz). Output 0 from the reconstruct block is offset -25 MHz from the [...]

My mistake, thank you. I appreciate your quick response.

closed 2023-02-16 by stevewillson